va3wam/hexBot #7: A pingable but non MQTT IP address crash loops code.

va3wam, 2021-09-01:
https://github.com/va3wam/hexBot/blob/b14a6327b650fce4c8ffc3d2039137bf4c722253/include/connectToMqttBroker.h#L14-L14

va3wam, 2021-09-01:
If you enter an address like 1.1.1.1 for the MQTT broker I address then the PING test passes which causes Hexbot too try and subscribe to the command topic. This appears to hang the processor which in short order causes the watch dog to issue a panic reboot. The chp reboots and repeats. For now the only work around is to comment out the line `bool tmp = connectToMqttBroker(); // Connect to MQTT broker.`  in main.cpp, reboot then use the web interface to ste a proper IP address. We need better defense against this hapening.

theAgingApprentice, 2022-04-17:
Not sure that we can fix this. For now we;'ll just call this a limitation. Will not close this ticket until the issue is documented in an FAQ.