va3wam/hexBot #15: Current draw of leg motors

va3wam, 2021-09-13:
Some of the motors tripped a reply in my power supply that indicated to me that they were drawing more current and/or voltage than other motors. I think that we should figure out how much current is going to be drawn by the combined leg motors.

nerdoug, 2021-09-13:
I remember measuring the current drawn by the servos, and noting that the
current drawn depends on the position of the servo. I'll see if I can find
those notes, and/or a power supply that makes it easy to measure current.

- Doug Elliott / VA3DAE / ***@***.***

nerdoug, 2021-12-15:
assigning to myself. Once I get the servos on my hexbot all working and calibrated, I'll use the SSD command to measure current drawn for various servo positions. (Looks like I need to replace a bunch of servos first.)

nerdoug, 2021-12-18:
should increase the size of the capacitor on the PWM board before doing current measurements.

theAgingApprentice, 2022-01-26:
The capacitors have been upgraded from 470uF to 1000uF and the motors have become more reliable. Closing this issue.

closed 2022-01-26 by theAgingApprentice