va3wam/hexBot #19: Figure out how to handle lots of leg movement simultaneously.

va3wam, 2021-09-18:
Use inverse kinematic library to try and control leg movements. Hopefully this will result in simultaneous moves.

nerdoug, 2021-11-30:
The flow approach to leg movement is based on moving each of the 18 servo motors in the legs, essentially at the same time, every 20 milliseconds. There is a challenges with this, because almost all of the leg movement computation is clustered at the 20 msec times.

-the 20 msec task to do the 18 servo changes (which I'll call the frame task (FT) ) needs to run every 20 msec. If other routines are running when the the next FT is due, it will be delayed until the other routines are done, and a check is made to see if it's FT time. Depending how the code in loop() is structured, there could be multiple routines running before the FT check.
To address this:
1) break long running routines into shorter (in terms of worst case execution time) routines that execute sequentially, allowing an FT check between routines
2) structure loop() so that at most 1 routine runs per loop() execution. This ensures the FT check at the start of loop() gets done after every routine execution. This is what we did in TWIPE.
3) use FREERTOS facilities to execute the FT according to a timer, and at a higher priority that other routines. This means that other routines will be paused when the timer expires, while the FT runs. This is the "pre-emptive" part of pre-emptive multitasking.

I'm proceeding to implement steps 1 and 2, with some metrics to see how we're doing at running a consistent 20 msec FT. If that can't be done, we can consider these steps:

a) slightly desynchronize the legs. Doing legs 1, 2 and three in one 20 msec slot, and legs 4, 5, and 6 in the next slot doubles the amount of time available for the leg movements. We could do one leg (3 servos) per 20 msec slot would give more time, and doing one servo per slot is the extreme, but might impact performance
b) start thinking about FREERTOS and pre-emptive multi-tasking.

One factor we'll need to examine, probably via performance metrics, is how much delay is caused by the PWN driver board's processing and I2C bus communications. We may need to make sure the 2 drivers are doing things in parallel as much as possible, and that we minimize using any operations that cause blocking. A last resort approach would be to build our own PWM driver that has a high speed connection to the ESP32, non-blocking communications, and PWM characteristics that are similar to the current board.

nerdoug, 2021-11-30:
one other mitigation if we can't keep up with 20 msec frame changes, would be an effort to optimize the code to reduce execution time. We have deliberately traded off execution speed and procedural efficiency for generality and highly structured data. Moving more towards real time and embedded coding techniques could reduce execution time significantly.

nerdoug, 2021-12-06:
git pull 52 implemented flows for 6 legs, which provides a way to control changing the 18 servos every 20 msec in order to move the legs in a manner described in a "flow". I think this addresses the base question in this issue, although fine tuning may be needed based on performance. closing this issue.

closed 2021-12-06 by nerdoug