va3wam/hexBot #20: Assemble Doug's robot.

va3wam, 2021-09-18:
Because the way the shin connects to the knee the direction to move the knee up and down is opposite for the left and right legs of the robot. We need to be sure that we coordinate our assembly so that our up and down agree with each other. Not sure of forward and backward are affected that same way at the moment.

nerdoug, 2021-12-15:
discovered that I was using a different convention for leg numbering, which is built in to some software. Will maintain this numbering until my bot is physically stable, and then do current consumption testing which will help Andrew to choose a bench power supply. Once that's done, I'll modify the software to match the original leg numbering scheme.

theAgingApprentice, 2021-12-16:
Switches are not installed but enough is done to allow us to start testing code  etc. Closing this issue.

closed 2021-12-16 by theAgingApprentice