va3wam/hexBot #38: Nail down universal coordinate values

theAgingApprentice, 2021-11-03:
Decisions:
1. We are going to use a "robot wide" coordinate system.
2. The x axis runs along the spine, the y axis runs across the shoulders, the z axis is up and down.
3. The Robot origin is in the centre of the robot chassis (x,y) and at the same height as the origin of the legs (z).
4. +x is forward. +z is up, +y is his left

To do:
Using Fusion360 create a diagram that shows  
1. Angle that the toe on each leg is from the origin x,y
2. The x,y,z home coordinate for each leg toe (home means 0 servo angles)

theAgingApprentice, 2021-11-03:
Home coordinates for each toe are:
Front left leg toe home coordinates: x=+18.63cm, y=+14.82cm,z=-10.56cm
Front right leg toe home coordinates: x=+18.63cm, y=-14.82cm,z=-10.56cm
Mid left leg toe home coordinates: x=0cm, y=+20.80cm,z=-10.56cm
Mid right leg toe home coordinates: x=0cm, y=-20.80cm,z=-10.56cm
Back left leg toe home coordinates: x=-18.63cm, y=+14.82cm,z=-10.56cm
Back right leg toe home coordinates: x=-18.63cm, y=-14.82cm,z=-10.56c

Angles for each leg are:
Front & Back Legs: 38.5d off x axis 
Mid Legs: 90d off  x axis 

Rough Diagram:
[image]
m

nerdoug, 2021-11-30:
home position coordinates given here are close to, but not quite the same as those in /docs/hex-coord-translations.ods, sheet "key locations" Not sure if the difference matters.

theAgingApprentice, 2022-04-17:
Moving to a new chassis and new numbers will be published. Can clode this issue.

closed 2022-04-17 by theAgingApprentice