va3wam/hexBot #39: Rework leg math

va3wam, 2021-11-03:
Switch the X Y and Z axis currently used by the lag math to the new Robot wide coordinate system per issue 38. Translate magic numbers to symbols for robot measurements.

nerdoug, 2021-12-02:
The mathematical complexity of using global coordinates in the leg math is prohibitive. Better (and feasible) to use a local coordinate system that aligns with the orientation of the leg. Supporting this is a routine that converts global coordinates to the local coordinates of a specific leg. Using similar axis names for local and global coordinates can be confusing, so this task will be implemented as reworking the leg math, comments, documentation and variable names to use local axes p, q and r consistently for all legs.

nerdoug, 2021-12-02:
oops. "close with comment" did more closing than I intended

theAgingApprentice, 2022-04-17:
Will be done as part of the reworking of the leg math that will be needed after we get the new chassis.