va3wam/hexBot #50: review robot measurements

nerdoug, 2021-12-01:
Our measurements for key points on the hexbot's body haven't been 100 percent consistent. Although we don't need a high degree of accuracy, we should make sure we are correct to the nearest .01 centimeter. THe reference used in the software are these arrays, defined in flows.h and populated in flows.cpp/setupFlows.

For completeness, we need to add one more measurement: vertical distance from center of hip servo to the origin.

theAgingApprentice, 2022-04-11:
I believe that this is now taken care of and these values are used in the movement calculations. Leaving for Doug to confirm and close.

theAgingApprentice, 2022-04-17:
Need to re-do once new chassis is built. Awaiting manufacturing process.