va3wam/hexBot #58: Investigate current and voltage getting to the motors

theAgingApprentice, 2021-12-08:
Initial voltage readings indicate that not enough power is getting to the motor controller boards.

nerdoug, 2021-12-15:
If we measure battery voltage, servo voltage and servo current, we'll need 3 analog to digital converters. ESP32 has 2 ADC controllers that support 18 ADC channels (with some restrictions) so we should be OK.

nerdoug, 2021-12-15:
I have a number of mini-voltmeter devices that can be hooked up to some of the unused duplicated pins on each PCA9685 to give a real time display of the voltage gong to the servo. Will provide a couple of these to Andrew.
NOTE: These run off the circuit they attach to, and will go blank if the voltage goes below their minimum operating requirement. They handle 4.5 - 5.0 fine, but I have seen them blank out.

theAgingApprentice, 2021-12-16:
Mini-voltage meters added on both bots. Closing issue.

closed 2021-12-16 by theAgingApprentice