va3wam/hexBot #59: Create MQTT command for testing servo currwnt draw

theAgingApprentice, 2021-12-08:
Create a command that puts Hexbot into a special test mode that allows us to put servos in known positions so that we can take current and voltage measurements. This code is not expected to be long term but is a special test situation to triage power issues that we are seeing (see issue #58 - investigate current and voltage getting to motors) .

nerdoug, 2021-12-09:
Added 3 new MQTT commands:
// SSD command = Set  Servo(s) to angle, expressed in degrees, zero is center of range
   if(cmd == "SSD")     // set servo(s) to degrees
   // command format: SSD, degrees, servo#, (optional end of range servo#)
   //    -90 <= degrees <= 90
   //    1 <= servo# <= 18
   //    servo #1 is hip for leg 1, the front right one

// SSP command - Set Servo(s) to PWM value
   if(cmd == "SSP")     // set servo(s) to PWM value
   // command format: SSP, PWM, servo#, (optional end of range servo#)
   //    110 <= PWM <= 490
   //    1 <= servo# <= 18
   //    servo #1 is hip for leg 1, the front right one

   // SSO command - Set Servo(s) to "OFF"
   if(cmd == "SSO")     // set servo(s) to OFF, where servo goes limp and draws no current
   // command format: SSO, servo#, (optional end of range servo#)
   //    1 <= servo# <= 18
   //    servo #1 is hip for leg 1, the front right one

Code is untested, since my hexbot needs some wiring, but I'm about to do a git merge / pull.

nerdoug, 2021-12-09:
Examples:

set all servos to center position = 0 degrees
   SSD,0,1,18
set all servos on right side to 45 degrees
   SSD,45,1,9
set left side knees to -30 degrees
   SSD,-30,11
   SSD,-30,14
   SSD,-30,17

set all servos on 2 front legs to PWM = 300
   SSP,300,1,3
   SSP,300,10,12

turn off all servos
    SSO,1,18
turn off leg 1's knee servo
    SSO,2

PR #60 (2021-12-09): Add SSD, SSP and SSO cmds for servo testing -
Fixes #59 add MQTT commands:
SSD - set a servo reange to an angle in degrees
SSP - set a servo range to a PWM value
SSO - set a servo range to "OFF" state
commit 361e0b17 (2021-12-09): Add SSD, SSP and SSO cmds for servo testing -
Fixes #59
closed 2021-12-09 by nerdoug