va3wam/hexBot #66: Issue random twitching of legs with no known changes in environment

nerdoug, 2021-12-16:
When the bot is sitting idle after some sort of servo movement, it will occasionally go into twitching spells, where arbitrary legs make small repeated moves with apparently random timing

nerdoug, 2021-12-16:
Some possibilities to ponder:
-I2C bus issues
    - pull up resistor issues
    - voltage or current variations
 - some flakey component
 - some flakey wiring

nerdoug, 2021-12-18:
Some troubleshooting techniques:
- increase 5V cap and see if twitching goes away.
- scope the I2C bus to see if there's activity when we're not driving servos, but they're twitching
- similarly, scope the +5 and +3.3 volts lines in various places
- scope various grounds signals, to see if they agree
- might be interesting to scope +5 / Gnd on a servo we're not driving at all.
- see if there's any twitching when only one PWM board is connected. If not, it might be a contention issue
- implement a second I2C bus so activity on both PWM boards can overlap without I2C competition

nerdoug, 2021-12-19:
Haven't seen this twitching (except on initial startup) since increasing the caps on the PWM boards. If this remains the case, I'll close this issue, blaming the caps. 

suspect the startup case can be solved by making sure the software sends some command to each servo before the 5V power is enabled, so the servo position isn't undefined when +5V appears. This could be done by the startup software if we put in a relay that controls the 5V power as we considered, and use pullups to ensure it comes up with 5V disabled until a command has been sent to all servos.

theAgingApprentice, 2022-04-11:
I have also noted the issue is much better with new caps.

theAgingApprentice, 2022-04-17:
We put larger capacitors on the controller boards and this seems to have greatly improved the issue. We are now discussing possibly making the caps  even larger to deal with boot up issues but we can consider this issue closed.

closed 2022-04-17 by theAgingApprentice