va3wam/hexBot #67: Slow down flow logic

theAgingApprentice, 2021-12-16:
Issue seen is that the flow movements are jerky and unpredictable. The suspicion is that this is timing related because when we increased the time interval for the SD commands from 20ms to 30ms the legs became more reliable. 

To do:
1. Figure out why extending the SD command from 20ms to 30ms fixed the SD execution.
2. Come up with a plan based on this knowledge to fix the flow executions to make them more reliable.

nerdoug, 2021-12-18:
This may be related to the current surge when a lot of servos move at once, and the standard capacitor that's pre-installed being too small. Opening another issue for capacitor sizing, but it may resolve this one too.

Note that the current amount of delays in the SSD command isn't workable in flows, because there are 18 servos involved in continuous motion, and the delays add up to a total delay that impacts control of the legs.

nerdoug, 2022-01-02:
I have to retract comment that larger delays in grouped servo commands aren't feasible. We could make this work by  increasing the time per frame from current default of 20 msec to something as big as 50 msec, or 20 frames per second. We would still have a cascading leg motion, which is undesirable, but we could have larger delays between servo commands. I think this is more masking the problem than solving the root cause.

nerdoug, 2022-01-12:
jerky leg movements have been resolved by a variety of improvements including:
- soldering connections to PWM boards to avoid vibration induced connection inconsistency
- changing servo movement frame rate from 50 FPS to 20 FPS
- increasing the capacitor on the PWM boards from 470 to 1000 microFarads

There are currently no explicit delays on the flow processing's servo movements. Adding any would be undesirable, because it wastes CPU that can be used for other purposes.