va3wam/hexBot #68: Create "push up" movement.

theAgingApprentice, 2021-12-16:


nerdoug, 2021-12-18:
As a starting point, here a MQTT.fx script I'm testing that moves the hexbot to a squatting position where the toes are 3.5 cm higher than the home position, i.e. the bot's body is 3.5 cm closer to the ground:

	// clear any previous flow
	send("fg,0,0")
	
	// go to the lying on belly position, with legs 3.5 cm above home position
	// since this is the first flow row, we jump to it directly, then pause for 340 msec
	send("fl,300,3,10,0,0,0, 0,0,3.5, 0,0,3.5, 0,0,3.5, 0,0,3.5, 0,0,3.5, 0,0,3.5")
	//
	// now that we've defined the flow, execute it
	send("fg,1,20,1")          // start er up, 20 msec, move toes and don't print coords for each frame

The arguments of the flow ("fl") command are:
300 - number of milleseconds to execute this move. first position is a special case, and this is ignored
3  - operating mode is relative to home position. numbers given for each leg are toe positions
      expressed as offsets in global X, Y, and Z directions from that leg's home position
10 - we're moving is straight lines between successive flow row toe positions
0,0,0  - parameters that would be used if we were following a parabola or ellipse rather a line between positions
0,0,3.5, - position for leg 1: 
                desired X coord is this leg's home position's X coord plus 0
                desired Y coord is this leg's home position's Y coord plus 0
                desired Z coord is this leg's home position's Z coord plus 3.5 cm 
0,0,3.5, - position for leg 2, as above for leg 1
0,0,3.5, - position for leg 3, as above for leg 1
0,0,3.5, - position for leg 4, as above for leg 1
0,0,3.5, - position for leg 5, as above for leg 1
0,0,3.5, - position for leg 6, as above for leg 1

nerdoug, 2021-12-18:
Here the full script (untested) for a full push up cycle:

```
var Thread = Java.type("java.lang.Thread");

function execute() {
    out("Hexapod Pushup Test.  File= 01__hexbot-pushup.js");
	
    myTWIPe = "Hexbot3C61054ADD98/commands"     // Doug's bot
//	myTWIPe = "calServoCC50E394F048/commands"     // Andrew's bot

    out(" ");
	
	// clear any previous flow
	send("fg,0,0")
	
	// go to the lying on belly position, with legs 3.5 cm above home position
	// since this is the first flow row, we jump to it directly, then pause for 340 msec
        // note that the safety box for each toe currently limits it to moving at most 6 cm
        //       away from the home position along any axis
	send("fl,300,3,10,0,0,0, 0,0,3.5, 0,0,3.5, 0,0,3.5, 0,0,3.5, 0,0,3.5, 0,0,3.5")
	//
	// go to the neutral or home position, slowly, over 3 seconds. this raises bot 3.5 cm higher
	send("fl,3000,3,10,0,0,0, 0,0,0,  0,0,0, 0,0,0, 0,0,0, 0,0,0, 0,0,0")
	//
	// stretch legs further, to 3.5 cm below the home position. This raises bot another 3.5 cm
	send("fl,3000,3,10,0,0,0, 0,0,-3.5, 0,0,-3.5, 0,0,-3.5, 0,0,-3.5, 0,0,-3.5, 0,0,-3.5")
	//
	// and go back to home position, which lowers the bot by 3 cm
	send("fl,3000,3,10,0,0,0, 0,0,0,  0,0,0, 0,0,0, 0,0,0, 0,0,0, 0,0,0")
	
	// now that we've defined the flow, execute it
	send("fg,1,20,1")    // start er up, 20 msec, move toes and don't print coords for each frame

    action.setExitCode(0);
    action.setResultText("done.");
    out("Parameter parameter set up complete");
    return action;
}
function send(cmd){
	out(cmd);
	mqttManager.publish(myTWIPe, cmd);
	Thread.sleep(500);
}
function out(message){
     output.print(message);
}
```

nerdoug, 2022-01-13:
the MQTT.fx script 01__hexbot-deep-up-c4.js is a demo pushup script. It does 6 and 3 legged pushups, with increased stability due to pulling the toes 4 cm closer to the center of the bot.

can't attach a .js file, so I'll try zipping it.
[01__hexbot-deep-pu-c4.zip](https://github.com/va3wam/hexBot/files/7860052/01__hexbot-deep-pu-c4.zip)

closed 2022-01-13 by nerdoug