va3wam/hexBot #74: Determine max movement for leg joints

theAgingApprentice, 2021-12-16:


nerdoug, 2021-12-18:
I haven't tested the new MOVE_LEG command, but once I do, it will be a way to determine the limits for each leg. Will probably need a special mode that bypasses the toe safety box mechanism so it doesn't keep the toe from getting close to boundary conditions we're looking for.

Still need a healthy safety margin to take imperfectly calibrated servo horn positions into account.

nerdoug, 2022-02-01:
There are a couple of constraints on leg movements:
-limits imposed by servo motor hardware. I think we're assuming plus and minus 90 degrees from center position, even though specs may allow more
-limits imposed by needing to avoid leg parts colliding with robot body. This can be determined by doing a lot of careful testing. The result is a set of pre-determined restrictions that don't change at run time
-limits imposed by needing to avoid hitting adjacent legs. This can be approximated by a conservative volume of allowable toe positions for each leg, but this sacrifices some safe movements. The toe safety box that's already implemented follows this approach, and is pre-determined and doesn't change at run time. A more ideal situation would be to track adjacent leg positions, and allow any movement that did not put any of the moving leg's parts within a settable safe distance from any part of either adjacent leg. This is NOT pre-determined, and must be done at run time.

theAgingApprentice, 2022-04-17:
Do this once we move to the HiTec servo motors. Rough centre is 300 PWM. Will account for fine adjustments in software with newer code that we have planned.