va3wam/hexBot #92: rewrite MQTT MOVE_LEG command, which has been disabled

nerdoug, 2022-01-07:
```
   // todo
```

https://github.com/va3wam/hexBot/blob/6a981e96352f3115a41f6d7aa0a0d8ca7432f8c7/include/mqttBroker.cpp#L440

nerdoug, 2022-01-07:
if (cmd == "MOVE_LEGx" || cmd == "MLx")   // the "x" additions effectively disable this command
   { // command format: ML, f_operation, leg, X, Y, Z
   // *************************** this command needs to be reworked to:
   // todo #92
   // allow all movement methods used in flow rows
   // use changed format of globCoordsToLocal routine

see issue 86 as well.

theAgingApprentice, 2022-04-17:
We have decided to remove this code (which is commented out anyhow) and not bother trying to fix it as we never use it anyhow.