va3wam/hexBot #141: Document VSC procedure to do a safe Github merge

nerdoug, 2022-02-07:
When multiple people are doing concurrent development, you need to make sure your origin merges use the latest code, and not the snapshot you took when you started your branch.

We have procedures documented to do this using the a combination of the Github browser interface and a terminal console:
      https://va3wam.github.io/versionControl/
but not one within the VSC environment, which has become our standard.

According to a website I found:
   BEFORE VS Code 1.48, VS Code Source Control has no UI for merging branches.
   reference: https://code.visualstudio.com/updates/v1_48#_git-overflow-menu-cleanup

However you can now do this, and I think the following steps are equivalent:

- make sure all the changes in your branch are staged and committed locally
- depending on your confidence level and experience with git, make backups of your changed source files
- click on branch name at bottom left of VSC window, and checkout the local main or master branch
- in source control, click the ellipsis at the top, then on "pull" on the resulting popup
- click on the branch name (main or master) in the bottom left and select your branch
- in source control, click the ellipsis, then on "checkout to", then your branch name
- in source control, click the ellipsis, then "Branch" then "Merge Branch", then main or master
- ( we could push the updated branch to origin, but that will be done for us as part of Pull Request)
- in source control, click the complicated icon to the right of the check mark to initiate a pull request
- enter a brief description, and click on the blue "Create" button
- you may get a message saying there's no upstream branch mirroring yours. That's because we were too lazy to publish above. Click the "Publish branch" button to have it done for you.
- A Pull Request form will appear at the top of the right VSC pane. If the "Close Pull Request" button isn't visible at the bottom, lower the pane boundary until you can see it all.
- ideally, the branch has no conflicts with the base branch. Otherwise, you need to fix that first.
- add comments in the "Leave a comment" box near the bottom if you wish
- click on the blue "Merge Pull Request" button
- click on the blue "Create Merge Commit" button
- click on the "Delete branch..." button
- at the top, click the blue "OK" to clean up both local and remote copies of the branch
- close the Pull request window
- in the bottom left corner, click the circle-with-arrows icon just to the right of "main" or "master", which will update the copy on your PC
- click OK on the popup window warning that this will push and pull commits
- if you're prone to making source changes before you create a branch, prevent that by opening a "I won't use this" branch now,
    which you can delete when you remember to open a branch properly. I call these "safety-n".
- if you're sceptical, use your browser to go to git and verify that your changes are there

I have used the above procedure successfully.

nerdoug, 2022-02-09:
guess we need to do overlapping Github uploads to test this before I can close the ticket.

va3wam, 2022-02-09:
OK, we can do that when we meet up own Sunday.  You seem to be making great progress on all of this!