va3wam/hexBot #157: Design servo motor controller power management circuit

theAgingApprentice, 2022-03-27:
Design a circuit that will allow the ESP32 MCU to control the power up sequence of the servo motor controller. The steps in the power-up sequence will (roughly) 1. ESP32 does boot up process, 2. Provide 3.3VDC to the PCA9685 processor on the controller board, and 3. Provide power to the 5DVC to the servo motors. The method of providing power to the servo motors must be able to handle 10AMP peak power and normal amps of around 3 AMPs.

nerdoug, 2022-03-28:
[Power-control.odg](https://github.com/va3wam/hexBot/files/8358860/Power-control.odg)

Attached a diagram I did a while ago to illustrate one approach to power management. It's also in the
hexBot/docs/circuitry folder.

nerdoug, 2022-03-28:
need to include pull-up an pull-down resistors to ensure that any signal that's floating because it hasn't been set up yet is pulled in the right direction.

theAgingApprentice, 2022-04-03:
Please add control for 3.3VDC from the ESP32 to the PCA9685 servo controller so that we can control the order that the servo controller CPU starts up and the motor power engages.

nerdoug, 2022-04-09:
I have somewhere between 3 and 6 LH1085 chips, which are 6 pin DIP solid state relays  that can control 3.3 VDC to PCA9685's nicely. have updated the diagram in the docs/circuitry/Power-control.odg file.

theAgingApprentice, 2022-04-10:
For 5VDC this may be an option: I have been thinking about the power control for our servo motor controllers I think we can very easily control the 3.3 and 5.0 VDC voltages for the PCA96856 servo drivers using a couple of Logic Level Enhancement-Type N-Channel MOSFET transistors. 

What I think that we need
1. A MOSFET with a Vgs of 3 VDC (or less) so that the ESP32 can turn it on and off. 
2. We are only using this as an on/off switch, not PWM, so the rise and fall time characteristics are not that important to us.
3. We want a transistor that is comfortable working at a constant current of say 3 AMPS 
4. We think that peak current could be 18 AMPs 

Type of MOSFET 
Check out this article: https://circuitjournal.com/which-mosfet-should-you-use-with-arduino
We can pull the pin low to a comma ground to ensure no floating problems and that the gate is closed at boot up.

Which MOSFET
According to this article: https://www.instructables.com/ESP32-and-DC-Motors-Tutorial-Part-1/, theRFP30N06LE MOSFET (spec sheet: https://www.sparkfun.com/datasheets/Components/General/RFP30N06LE.pdf) can be driven by an ESP32. 

Where to buy
A pack of 5 of these can be purchased on Amazon Canada for $11.00 ([https://www.amazon.ca/Gikfun-RFP30N06LE-N-Channel-Mosfet-Arduino/dp/B01GNORMWA/ref=sr_1_1_sspa?crid=L6WB2ZMBG4G2&keywords=RFP30N06LE&qid=1649024246&sprefix=rfp30n06le%2Caps%2C175&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyVERDWkY4MEVIRkJQJmVuY3J5cHRlZElkPUEwNTUxNDkzMzVEUlFWSUdIWlowTyZlbmNyeXB0ZWRBZElkPUEwNDQ5MzE1M1VFRjFWSDBXMUNVNCZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=](https://www.amazon.ca/Gikfun-RFP30N06LE-N-Channel-Mosfet-Arduino/dp/B01GNORMWA/ref=sr_1_1_sspa?crid=L6WB2ZMBG4G2&keywords=RFP30N06LE&qid=1649024246&sprefix=rfp30n06le,aps,175&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyVERDWkY4MEVIRkJQJmVuY3J5cHRlZElkPUEwNTUxNDkzMzVEUlFWSUdIWlowTyZlbmNyeXB0ZWRBZElkPUEwNDQ5MzE1M1VFRjFWSDBXMUNVNCZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=))

If you are up for it check my thinking and let me know if you feel that these are worth purchasing.

theAgingApprentice, 2022-04-11:
I ordered the MOSFETs as well as some extra PCA96856 boards so that we can prototype this circuit.

theAgingApprentice, 2022-04-17:
Another requirement here is to isolate back EMF from the motors so it doe snot interfere with the ICs.

theAgingApprentice, 2022-04-17:
Note that Doug's diagram in the docs/circuitry needs to be updated further but it is the ongoing requirement ref doc and design for this ticket.

nerdoug, 2022-05-23:
Created an initial rough draft of GPIO pin assignments for the functions we can foresee at this point. The actual GPIO choices are likely to change in the board layout process to simplify connections, but this does show that we have enough GPIO pins available, considering their various restrictions, to achieve the current plan, we a few leftover pins.

Notes:
-this assignment considers restrictions on some digit pins that can only be input
- whether the pin coms up with weak pull-up, weak pull-down, or neither was NOT considered
- we avoided using A/D controller 2 because it conflicts with WiFi in some way

Assignment file location is  ..docs/circuitry/GPIO-pins.ods

nerdoug, 2022-05-29:
reference for microprocessor interfacing: https://w9xt.com/page_microdesign_toc.html

nerdoug, 2022-05-30:
updated docs/circuitry/Power-control-2.odg, and GPIO-pins.ods to reflect:
-monitoring 5V voltage at only one point, rather than for each PWM controller
-having both a digital enable signal, and an analog sense signal for the common 5V voltage monitor
-moved 5V current sensors to show how they need to be in series with measured circuit
-used different lines for digital and analog GPIO connections
-added a table listing possible chips for various functions
-added 5V and 3.3V LEDs
-rearranged GPIO pin usage a bit, but ended up using the exact same number of pins in total

nerdoug, 2022-06-17:
current power design is reflected in these block diagrams in ..docs/circuitry/ :

Power-control-VN7050ASTR.odg (reflects use of 7050 to control 25V voltage measurement circuit)
power-test-config.odg (includes voltage divider resistor values for measurement circuits)
voltage-dividers.odg (includes the math used to derive resistor values)

nerdoug, 2022-09-25:
Updated power-test-config.odg to reflect GPIO control better.