va3wam/hexBot #158: Confirm that our robots are wired the same way

theAgingApprentice, 2022-03-28:
Recent testing showed a difference in behaviour between Doug and Andrew's robot. Create a script that moves each leg one at a time and compare the two robots to see if they behave the same. Note that this issue coincided with a voltage regulator failing in Andrew's robot. We expect the script and test to prove that wiring is not the issue.

nerdoug, 2022-03-31:
compiler issues 14 and 15 include a script called testAllLegs.txt / 01__hb-testAllLegs.js  which moves each servo for each leg in succession.

theAgingApprentice, 2022-04-10:
Script shows we are wired the same way but there are issues with motor behaviour. Other tickets track the new issue and this ticket can be closed.

closed 2022-04-10 by theAgingApprentice