va3wam/hexBot #160: Doug's servo motors are unhealthy

theAgingApprentice, 2022-04-03:
A complete survey of all 18 motors required with a strong suspicion that replacements are required. Also check out the motor controllers themselves.

nerdoug, 2022-04-11:
my leg 6 definitely has problems. makes a grating sound when you move servo manually, and occasionally jumps unpredictably when moving under program control.

The test all legs scripts doesn't do a very good job of testing. all legs seem to perform OK when it's run, including leg 6. I added coded to dump servo commands and timing, and it all looks right. I'll put it in another disposable issue to keep this one open.

Will write and compile a test leg 5 script, so I can compare a possibly good leg with a known bad one.

nerdoug, 2022-04-11:
added the debug code via issue 164, and followed the full procedure in issue 141, which worked exactly as documented

theAgingApprentice, 2022-04-17:
In addition to the problems documented above here are some other relevant notes. This is the same stuff added to the ticket tracking issues with motors on Andrew's robot. Doug has figured out that the script we originally used does not reveal the issue but with a modified script issues can be highlighted. This suggests that the motors may not be at fault so much as the way in which our code works. We have a few issues that we think are contributing to motor instability. 

1) The controllers can come up in funny states. We need better power control of the 3.3 and 5 VDC power to the controllers to address this.
2) While we did increase the capacitors on the controller boards from 450 uF to 1000 uF this still may not be enough. We suspect brown-outs during power up may contribute to the issues of motors starting up in a bad state of either being stiff or limp.
3) We have notice that the quality of the MG996R servo motors is not very good and there is noticeable differences between batches of motors hinting at an inconstant manufacturing process. We are going to switch to HiTec HS-485HB motors to see if that gets rid of some of the issues with "play" that we are seeing.

nerdoug, 2022-09-25:
As we change over to the new motors on a new chassis, the priority of this issue diminishes, or at least is reassessed with the new motors. If the issue is solely with old motors that are no longer used, it's probably not worth investigating.