va3wam/hexBot #162: Calibrate HS-485HB servos & add software adjustment

nerdoug, 2022-04-10:
We can't assume that the new HS-485HB servo motors act exactly the same as the old MG 996R servos. We will likely need to use different PWM numbers for a given rotation, and need to measure the new motors to see how we generate these. Hopefully, the new motors are as consistent as the old ones, and we can use a universal algorithm, with per servo calibration values. This will allow us to fine tune the positioning using a 24 position horn.

nerdoug, 2022-04-10:
new MQTT commands to allow testing of servos:

Servo_Test_PWM  <side (0-1)> <port(0-15)> <PWM (100-500?)>
Shortform: STP
Example: STP 0, 15, 300

Servo_Test_Angle <side (0-1)> <port(0-15)> <angle (-90-+90)>

The second command assumes that you have logic in place to translate angles to PWM, which won't be known right away.

nerdoug, 2022-04-11:
Will need to find some of the new bolts needed to attach new servo horns to leg parts. They're tiny. I found 4 in the right thread in my junk box, but none are long enough to do the job.

theAgingApprentice, 2022-04-11:
Purchased bolts on Amazon. Delivery should be in a couple of days.

nerdoug, 2022-05-12:
software adjustment of servo position was implemented and documented under issue #173, Add robot specific boot up logic. Can't close this issue until testing is done with new servos using new body parts that match mounting bolt positions.

commit 1a45e6af (2022-05-12): Calibrate HS-485HB servos & add software adjustment
Fixes #162
commit a4a57d7a (2022-05-12): Calibrate HS-485HB servos & add software adjustment
Fixes #162
commit 5494d7c4 (2022-05-12): Calibrate HS-485HB servos & add software adjustment
Fixes #162
closed 2022-05-12 by nerdoug