va3wam/hexBot #166: Create a new  robot chassis

theAgingApprentice, 2022-04-17:
As we switch from MG996R to HS-485HB hobby servo motors we will be making 2 main changes. 

The first change is that we will move the location of the toe in the neutral position from outside of the knee too directly below the knee. Testing has shown that this will put less stress on the servo motors.

The second change is that we are moving from an M3 metric screw standard to an imperial 2-56 screw standard. This means that our tapped holes now go from 3mm to 0.086 inches. We will be using 2 sizes of drills to accommodate this new screw standard.

1) When we want our 2-56 screws to pass through the part via non threaded holes we will use a 3/32" (0.09375") drill holes. An example of when we do this is the connection of the ankle servo horn to the lower shin.   
2) When we want our 2-56 screws to thread into a part we will use a 0.07" drill to make the hole. An example of when we do this is the connection of the foot to the ankle servo.

theAgingApprentice, 2022-04-18:
Have a new foot design and have saved the tool paths as a template in the cloud. Feeds are OK and finish seems passable.
[image]

nerdoug, 2022-04-18:
looks good. The foot looks shorter than the old one, or it this just my pathetic eyesight? Wondering if having the stub at the end a little longer would make it easier to attach "boots" is a stable way?

nerdoug, 2022-04-18:
Having the foot length variable depending on whether or not a boot is attached, and how thick it is, would complicate the math. Maybe we should adopt a standard boot thickness, and either never have him go barefoot, or accept inaccuracies if he does.

theAgingApprentice, 2022-04-19:
I think it must be the perspective of the model I choose to show the new design. The new foot is exactly the same length as the old one. This may be an issue now that I think of it. My plan is to machine shoes that will add to the foot length. I guess I can shorten the leg to account for the shoe. My plan is to porches rubber coating that we can dip the shoes into to get grippy feet.